Coopdis/easyMesh #1: error compiling; issue with SimpleList

mhdhgh, 2016-09-07:
amazing work you've done here, thank you.
but I keep getting errors compiling you're startHere. trying to read your source codes and can't solve the problem.
any help and insight will appreciated.

```
Arduino: 1.6.9 (Windows 10), Board: "Generic ESP8266 Module, 80 MHz, 40MHz, DIO, 115200, 512K (64K SPIFFS), ck, Disabled, None"

In file included from
...\Arduino\libraries\easyMesh\src\eashMeshConnection.cpp:18:0:

...\Arduino\libraries\easyMesh\src\easyMesh.h: In instantiation of 'T* SimpleList<T>::erase(SimpleList<T>::iterator) [with T = meshConnectionType; SimpleList<T>::iterator = meshConnectionType*]':

...\Arduino\libraries\easyMesh\src\eashMeshConnection.cpp:44:37:   required from here

...\Arduino\libraries\easyMesh\src\easyMesh.h:69:8: error: no matching function for call to 'SimpleList<String>::SimpleList(const SimpleList<String>&)'

 struct meshConnectionType {

        ^

...\Arduino\libraries\easyMesh\src\easyMesh.h:69:8: note: candidates are:

In file included from ...\Arduino\libraries\easyMesh\src\eashMeshConnection.cpp:11:0:

...\Arduino\libraries\SimpleList/SimpleList.h:39:5: note: SimpleList<T>::SimpleList(SimpleList<T>&) [with T = String]

     SimpleList(SimpleList& from)

     ^

...\Arduino\libraries\SimpleList/SimpleList.h:39:5: note:   no known conversion for argument 1 from 'const SimpleList<String>' to 'SimpleList<String>&'

...\Arduino\libraries\SimpleList/SimpleList.h:22:5: note: SimpleList<T>::SimpleList() [with T = String]

     SimpleList()

     ^

...\Arduino\libraries\SimpleList/SimpleList.h:22:5: note:   candidate expects 0 arguments, 1 provided

...\Arduino\libraries\SimpleList/SimpleList.h:51:17: error:   initializing argument 1 of 'SimpleList<T>& SimpleList<T>::operator=(SimpleList<T>) [with T = String]'

     SimpleList& operator=(SimpleList from)

                 ^

...\Arduino\libraries\SimpleList/SimpleList.h:144:33: note: synthesized method 'meshConnectionType& meshConnectionType::operator=(const meshConnectionType&)' first required here 

                     newArray[i] = _internalArray[i + 1];

                                 ^

exit status 1
Error compiling for board Generic ESP8266 Module.
Error downloading http://arduino.esp8266.com/stable/package_esp8266com_index.json
```

Coopdis, 2016-09-07:
Hi mhdhgh,

Thanks so much for giving my library a test run.

I'm afraid that I am not able to recreate this error on my end.  I am able to compile and run the startHere code perfectly well using Arduino 1.6.9 running on my Mac targeting Board "Generic ESP8266 Module".

Do you have the correct SimpleList library installed?  I got mine here...
https://github.com/blackhack/ArduLibraries/blob/master/SimpleList/SimpleList.h

It sits in a folder called "SimpleList" that is in my Arduino libraries directory.

Bill

RudyFiero, 2016-09-08:
I just downloaded the current library and the SimpleList.h from the above link. I get the same error.

I'm running Arduino 1.6.9 on a Windows 10 machine. Current stable package for ESP8266.

shinchan14, 2016-09-08:
have you downloaded library easywebserver and easywebsocket?
because they are connected with easymesh

RudyFiero, 2016-09-08:
Yes. I had to include it so I did.

Coopdis, 2016-09-08:
Interesting.

Perhaps this is a problem that occurs on Windows and not on Mac?  I don't have a copy of Windows 10 to debug this with, but I will try and compile on Linux tomorrow, maybe the same issue will crop up there?

re: easyWebServer and easyWebSocket
I don't believe that easyMesh is dependent on either of these libraries.  The startHere example should compile and run fine without these libraries.

The demoToy example does use these two libraries... so if you want to run demoToy you'll have to install easyWebServer and easyWebSocket.

RudyFiero, 2016-09-08:
Yes I installed all the required libraries for demoToy. Same error.

shinchan14, 2016-09-08:
I had the same problem but after installing all the libraries
starhere and demotoy can be used

Coopdis, 2016-09-08:
Great shinchan14!  What platform and IDE version are you using?

shinchan14, 2016-09-08:
i make Windows 7 
and arduino IDE version 1.6.7

shinchan14, 2016-09-08:
can i aks you?
I have 2 modes for the mash
how am I going to check if they are connected
if true should perogram running and release the results "hello form node"
isn't it?

  // if the time is ripe, send everyone a message!
  if ( sendMessageTime != 0 && sendMessageTime < mesh.getNodeTime() ){
    String msg = "Hello from node ";
    msg += mesh.getChipId();
    mesh.sendBroadcast( msg );
    sendMessageTime = 0;
  }

 my problem is "hello form node" not show

thanks
}

Coopdis, 2016-09-08:
Hmm...

Are you able to see the coordinated LED blinking?  The leds should be blinking twice when there are 2 nodes on the mesh.

Also, can you post the serial output that you are seeing?

shinchan14, 2016-09-08:
oh .. should use led,
I do not do it
ok I will try using LED
but can i detect nodes without any LED?

Coopdis, 2016-09-08:
Sure.

If you post the serial output I can tell you more about what is going on.

But for now... I'm off to bed.

shinchan14, 2016-09-08:
ok thanks for respon and your program,
i will try it again..
and share the result

Coopdis, 2016-09-08:
Shinchin,

Can we move your issue to a new thread?   Your issue is different and not related to the Windows compiling error.

Thanks

shinchan14, 2016-09-09:
sory

mhdhgh, 2016-09-09:
so I just tried compiling on a Linux machine and got the exact same error, anoybody get a successful build on Linux? weird how it compiles on Mac and Win7 but not on Win10 and Linux.
I also compiled SimpleList and ArduinoJson codes and examples without any problem, so maybe there's something in the way you're using SimpleList?
I'm still going through you're code to make something of the error...

Coopdis, 2016-09-14:
Hey guys.

Sorry for the pause there, I got crushed with work for a spell.

I've just installed Arduino on Linux and compiled the startHere code, and everything is working fine for me.  To be clear I installed Arduino 1.6.11 on a Debian 8 64-bit machine.  Things are compiling fine.  I should say that the Arduino IDE installed buy apt-get on my Debian box is some old janky POS.  You have to install 1.6.11 by downloading it yourself from the Arduino site.

Perhaps an upgrade from Arduino 1.6.8 to 1.6.11 is all that is needed?

I also tried to install Arduino 1.6.11 and 1.6.8 on an old Windows XP virtual machine that I use from time to time, but no luck.  Windows XP appears to be too old to run the newer Arduino IDEs.  If someone wants to contribute a Windows 10 installer, I'm happy to look into this problem on Windows 10, but I'm not likely to buy (or otherwise get) a Windows 10 copy on my own in the near future.

easyMesh on Debian 8 64bit + Arduino 1.6.11 is compiling and running for me.

Bill

bodgybrothers, 2016-09-19:
Just to throw something at this. I compiled on win 10 in Arduino 1.6.11. Originally I had an old version and upgraded Arduino and I got a compile error. I deleted all the libs and reinstalled everything fresh, updated the esp8266 in tools -> board manager. Compiled again and it worked. I also recieved some errors on the JSON libs the first time because they were simply too old. If all else fails, delete all and install a fresh batch. A simple upgrade should be enough though.

Coopdis, 2016-09-19:
bodgebrothers,

Thanks so much for tracking this down.  Solution... upgrade to the latest IDE.

closed 2016-09-19 by Coopdis

mhdhgh, 2016-09-22:
I had all the latest software and libraries; but deleting and reinstalling all seems to have solved the problem. thanks all