Coopdis/easyMesh #3: Serial.printf msg error

winduajiprasetiyo, 2016-09-10:
please help
i have 2 nodeMCU v1(esp-12e)
i try ur library "startHere.ino" and upload to each nodeMCU
but when im open serial monitor IDE print this

startHere: New Connection, adopt=1
startHere: Recive from 8821335 msg=D[]

im was change the string "Hello from node" to "blablabla" but the result is same

Coopdis, 2016-09-14:
Hi winduajiprasetiyo,

Sorry for the delayed response, I've been crushed with work.

Thanks for trying out easyMesh.

I'm not sure what is going wrong, but the first thing to do is to turn on more debugging messages.  In the startHere example, you can do that by uncommenting line 33 and commenting line 34.

`mesh.setDebugMsgTypes( ERROR | MESH_STATUS | CONNECTION | SYNC | COMMUNICATION | GENERAL | MSG_TYPES | REMOTE ); // all types on `

This should fill your serial line with a bunch of different messages.  Post the output here, and I will likely be able to tell you what is going on.

Bill

bodgybrothers, 2016-09-19:
I've noticed the same thing. The pointer to &msg seems to be messed up. However, it is correct in the library code... I compiled on both windows and ubuntu with the same result. I tried reading the code, but couldn't see why the msg is lost between the cpp library and the arduino callback event.

ENVIRONMENT:
compiled on Arduino 1.6.11
ESP-12E
ArduinoJson 5.6.6
simpleList Latest

LOG:

startHere: New Connection, adopt=1

0x20	meshRecvCb(): data={"dest":73584,"from":73584,"type":4,"msg":{"time":704838282,"num":9,"adopt":true}} fromId=73584
0x20	sendBroadcast(): msg=Hello from node 143179
0x20	broadcastMessage(): from=143179 type=8, msg=Hello from node 143179 exclude=NULL
0x20	sendMessage(conn): conn-chipId=73584 destId=73584 type=8 msg=Hello from node 143179
0x20	Sending to 73584-->{"dest":73584,"from":143179,"type":8,"msg":"Hello from node 143179"}<--
0x20	meshRecvCb(): data={"dest":143179,"from":73584,"type":8,"msg":"Hello from node 73584"} fromId=73584
0x20	broadcastMessage(): from=73584 type=8, msg=Hello from node 73584 exclude=73584
startHere: Received from 73584 msg=

0x20	sendBroadcast(): msg=Hello from node 143179
0x20	broadcastMessage(): from=143179 type=8, msg=Hello from node 143179 exclude=NULL
0x20	sendMessage(conn): conn-chipId=73584 destId=73584 type=8 msg=Hello from node 143179
0x20	Sending to 73584-->{"dest":73584,"from":143179,"type":8,"msg":"Hello from node 143179"}<--
0x20	sendMessage(conn): conn-chipId=73584 destId=73584 type=5 msg=[]
0x20	Sending to 73584-->{"dest":73584,"from":143179,"type":5,"subs":[]}<--
0x20	meshRecvCb(): data={"dest":73584,"from":73584,"type":6,"subs":[]} fromId=73584
0x20	sendBroadcast(): msg=Hello from node 143179
0x20	broadcastMessage(): from=143179 type=8, msg=Hello from node 143179 exclude=NULL
0x20	sendMessage(conn): conn-chipId=73584 destId=73584 type=8 msg=Hello from node 143179
0x20	Sending to 73584-->{"dest":73584,"from":143179,"type":8,"msg":"Hello from node 143179"}<--
0x20	sendBroadcast(): msg=Hello from node 143179
0x20	broadcastMessage(): from=143179 type=8, msg=Hello from node 143179 exclude=NULL
0x20	sendMessage(conn): conn-chipId=73584 destId=73584 type=8 msg=Hello from node 143179
0x20	Sending to 73584-->{"dest":73584,"from":143179,"type":8,"msg":"Hello from node 143179"}<--
0x20	sendMessage(conn): conn-chipId=73584 destId=73584 type=5 msg=[]
0x20	Sending to 73584-->{"dest":73584,"from":143179,"type":5,"subs":[]}<--
0x20	meshRecvCb(): data={"dest":73584,"from":73584,"type":6,"subs":[]} fromId=73584
0x20	meshRecvCb(): data={"dest":143179,"from":73584,"type":8,"msg":"Hello from node 73584"} fromId=73584
0x20	broadcastMessage(): from=73584 type=8, msg=Hello from node 73584 exclude=73584
startHere: Received from 73584 msg=

bodgybrothers, 2016-09-21:
Ok, false alarm. I just went back to this and saw it right away.. 
the "startHere.ino" has a typeo. msg needs to be a char pointer so needs msg.c_str() 

This:
```
void receivedCallback( uint32_t from, String &msg ) {
  Serial.printf("startHere: Received from %d msg=%s\n\r");
}
```

Should be:
```
void receivedCallback( uint32_t from, String &msg ) {
  Serial.printf("startHere: Received from %d msg=%s\n\r", from, msg.c_str());
}
```

Coopdis, 2016-09-21:
Great! 

Thanks for looking into this.  i've just pushed this change to github.

Bill

closed 2016-09-21 by Coopdis