Coopdis/easyMesh #4: information about easy mesh

Palagoa, 2016-09-15:
i just download and try you easy mesh exemples and it work fine in my 10 devices that i have covering the distance more the 5 KM, its a great work that you did here.

but i have a question about this, this code is possible to use with the Lua Code?
if not how can i use it to use the esp8266 gpio with this coding? you have any exemple where you use the mesh and the gpio of the esp8266, making a input in one of them that makes a ouput in other one.

best regards

Coopdis, 2016-09-15:
Palagoa,

That is great to hear!  I would love to know more about your system and what you are using it for.

I have no experience with Lua, so I can't say if the library is Lua compatible or how you might use it with Lua.

What I can tell you is that you can do GPIO quite easily in 2 ways that I do have experience with.

1) the standard Arduino GPIO routines all work great.  If you aren't familiar, google "arduino digitalRead" to get started.  digitalWrite and analogRead also work great.

2) The standard esp8266 GPIO routines also work perfectly well, though they are not as easy to use.  Check out Kolban's book [http://neilkolban.com/tech/esp8266/](url) for a primer.

Bill

closed 2016-09-15 by Coopdis

Palagoa, 2016-09-16:
coopdis,

Regarding the network that i am trying to build its to get inpluses from energy, water , gas counter, from temperature probes, and compile every thing online to get track of usage of resources.
something like this https://thingspeak.com/channels/45572 but i have around 150 different inputs, and i want to try to connect 100 esp8266  http://www.banggood.com/Geekcreit-Doit-NodeMcu-Lua-ESP8266-ESP-12E-WIFI-Development-Board-p-985891.html

Now the issues :(

i not a programer but i  can read an modify all the code in order to get what i want if i have a good examples,

regarding the programming of the arduino a have a couple of experience to do simple programs but my intention is only to use the esp.

regarding the kolban book i will read tommorow about it.

best regards 
Paulo

durvesh09, 2017-07-10:
Hi,
   how to use raspberry pi as one of the node in the esp8266 mesh network ?