Coopdis/easyMesh #5: What happens if the first node (that created the AP) gets turned off?

sfranzyshen, 2016-09-17:
what happens if the first node (that created the AP) gets turned off? Do the nodes recover and do wifi connected device reconnect?

Coopdis, 2016-09-18:
All of the nodes create an AP.  All of the nodes also act as an STA.  If a node (I'll call this node off-node) is turned off, what ever links it has established (either as an AP or as a STA ) are broken.  All of the nodes that were connected to off-node as STAs will immediately begin connecting with other nodes that are still on.  The mesh will automatically will repair/restructure the network without the off-node.  If the off-node turns back on, then it will join the network in the usual way.

Regarding other WiFi connected devices... the way that they connect is not controlled by easyMesh, so how they behave is up to your application.

closed 2016-09-18 by Coopdis

sfranzyshen, 2016-09-19:
Thank you for your response ... I still have more questions ... I started a separate discussion topic over at [esp8266.com ](http://www.esp8266.com/viewtopic.php?f=6&t=11849)