Coopdis/easyMesh #7: Porting easyMesh to other platforms

sfranzyshen, 2016-09-25:
Since this concept is based on basic wifi hardware and standards ... what would prevent the protocol from being ported to other platforms ... such as linux(openwrt), android, or other arduino wifi libraries?

Coopdis, 2016-09-25:
Yes.  This should be pretty straight forward.

I'd love to see a linux port that can create a bridge to regular ip networks.

An Arduino port is less attractive to me (personally).  For the moment, if you want to run Arduino and use Wifi, then esp8266 is a pretty obvious choice.  This is likely to change, but who knows exactly how? 

When I first started this project I used the regular old Arduino Wifi stack for esp8266.  It was so easy!  I love Arduino for that.  But I could not squeeze the kind of performance that I was looking for out of the Arduino stack.  Latencies were really bad and unpredictable and I wasn't able to get anything even close to realtime.  So I moved to the esp8266 SDK.  The advantage of the SDK is that it runs way (way!) faster than the Arduino wifi stack.

sfranzyshen, 2016-10-07:
I started a separate discussion topic over at [esp8266.com ](http://www.esp8266.com/viewtopic.php?f=6&t=11849)