Coopdis/easyMesh #8: Json object destroying data argument.

muratdemirtas, 2016-09-25:
Hello dude. This is very good library for dynamic mesh. Thx you. But i found a bug in EasyMeshConnection:: 286. Received arguman data is destroying after json parsing. With this problem adhoc messaging doesnt work. i created a new copy of data argument before json parsing. And looks like i solved.

////////////////////After data argument is corrupting//////////////////////

        case SINGLE:
          { // bla bla
                String tempStr( data );
                staticThis->sendPackage( staticThis->findConnection( (uint32_t)root["dest"] ), tempStr );
            }
Tempstr is parsing corrupted data and sending to another node of mesh. İn this case Adhoc messaging is doesnt work. I spotted.

/////////////////////////////Corrupted data received from destination node/////////////////////////////
0x1	meshRecvCb: parseObject() failed. data=dest<--      
0x200	meshRecvCb(): data={"dest":23356,"from":1515899,"type":5,"subs":[{"chipId":1859811,"subs":[]}]} fromId=1515899
0x200	Recvd from 1515899-->dest<--                           //corrupted data.

And i created copy of the data and problem solved.

    String kopya(data);      //copy data before json parsing

    DynamicJsonBuffer jsonBuffer( JSON_BUFSIZE );
    JsonObject& root = jsonBuffer.parseObject( data );
    if (!root.success()) {   // Test if parsing succeeded.
        staticThis->debugMsg( ERROR, "meshRecvCb: parseObject() failed. data=%s<--\n", data);
        return;
    }

//////////////////////////Adhoc side serial messages/////////////////////////
0x200	Mesaj: Bu mesaj 1859811 'den 23356 icin gonderildi. Aktarimdayim.
Gonderilen veri dest                       //using String tempStr(data)
Gonderilen veri2{"dest":23356,"from":1859811,"type":9,"msg":"hello"}    //using String kopya(data); before json

///////////////////////Destination nodemcu side serial message/////////////

0x200	receivedCallback():
0x200	Mesaj: This message for us..

Sorry for my bad english.

sholehhudin, 2016-10-20:
Thanks @muratdemirtas . this help me alot

sfranzyshen, 2016-10-20:
Can anyone explain why this works?

RudyFiero, 2016-10-21:
I am still having problems fully understanding "this" in code. Or do you mean a different "this". ;)

To be honest, all of this is above my head. But I am certainly willing to test this once you have something you want tested with more units. 

Right now my interest is in being able to connect to this network with a tablet, seamlessly. Much as you have described. I'm working on understanding more of the pieces to make that work, without getting too involved in the lower level code. Because this drives me crazy. :)

sfranzyshen, 2016-10-21:
I meant why the fix @muratdemirtas posted works? just creating a String of data fixes data structure some how?

BlackEdder, 2016-10-21:
> Right now my interest is in being able to connect to this network with a tablet, seamlessly. Much as you have described. I'm working on understanding more of the pieces to make that work, without getting too involved in the lower level code. Because this drives me crazy. :)

@RudyFiero I implemented a basic client (I want to use it on a raspberrypi) that only does the minimal needed to receive messages: https://github.com/BlackEdder/easyMesheD That might be a better starting point to understand how to connect a tablet.

muratdemirtas, 2016-10-21:
@sfranzyshen because , memory leak in arduino json library.

closed 2022-06-26 by muratdemirtas