Coopdis/easyMesh #11: Power usage

BlackEdder, 2016-10-09:
More asking for information than opening an issue, but any clue how much the power usage is of this setup? Say I wanted to run on batteries how long could I expect them to last?

Is it possible to setup a low power network, where nodes go in and out of sleep?

sfranzyshen, 2016-10-09:
This question might be better handled by the esp8266 forum ...  http://www.esp8266.com/

as far as nodes going to sleep ... that would probably mess things up

RudyFiero, 2016-10-09:
Issue should really be for code issues only. Feature requests, support, that kind of stuff belongs ... 
OK I don't know. 

So the esp8266 site is a reasonable option.

sfranzyshen, 2016-10-09:
sorry ... by all means open an issue to request sleeping functionality ... but a generic question about how much power does your code use ... seams like a hardware question  ...

closed 2016-10-09 by BlackEdder