Coopdis/easyMesh #14: Can not Compile examples

hubshi, 2016-10-26:
I am using ESP8266 ESP-12 , Arduino Version 1.6.11, ESP8266 Board Package Version 2.3.0. Following is the Error Log.

Arduino: 1.6.11 (Windows 7), Board: "Generic ESP8266 Module, 80 MHz, 40MHz, DIO, 115200, 4M (1M SPIFFS), nodemcu, Disabled, None"

C:\Development\Embedded\Arduino\Software\arduino-1.6.11\arduino-builder -dump-prefs -logger=machine -hardware C:\Development\Embedded\Arduino\Software\arduino-1.6.11\hardware -hardware C:\Users\User\AppData\Local\Arduino15\packages -hardware C:\Users\User\Documents\Arduino\hardware -tools C:\Development\Embedded\Arduino\Software\arduino-1.6.11\tools-builder -tools C:\Development\Embedded\Arduino\Software\arduino-1.6.11\hardware\tools\avr -tools C:\Users\User\AppData\Local\Arduino15\packages -built-in-libraries C:\Development\Embedded\Arduino\Software\arduino-1.6.11\libraries -libraries C:\Users\User\Documents\Arduino\libraries -fqbn=esp8266:esp8266:generic:CpuFrequency=80,FlashFreq=40,FlashMode=dio,UploadSpeed=115200,FlashSize=4M1M,ResetMethod=nodemcu,Debug=Disabled,DebugLevel=None____ -ide-version=10611 -build-path C:\Users\User\AppData\Local\Temp\buildb8f9fbd0161026bef8078c61b5564ac2.tmp -warnings=all -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool.path=C:\Users\User\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.9 -prefs=runtime.tools.mkspiffs.path=C:\Users\User\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\0.1.2 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\User\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -verbose C:\Users\User\Documents\Arduino\libraries\easyMesh\examples\demoToy\demoToy.ino
C:\Development\Embedded\Arduino\Software\arduino-1.6.11\arduino-builder -compile -logger=machine -hardware C:\Development\Embedded\Arduino\Software\arduino-1.6.11\hardware -hardware C:\Users\User\AppData\Local\Arduino15\packages -hardware C:\Users\User\Documents\Arduino\hardware -tools C:\Development\Embedded\Arduino\Software\arduino-1.6.11\tools-builder -tools C:\Development\Embedded\Arduino\Software\arduino-1.6.11\hardware\tools\avr -tools C:\Users\User\AppData\Local\Arduino15\packages -built-in-libraries C:\Development\Embedded\Arduino\Software\arduino-1.6.11\libraries -libraries C:\Users\User\Documents\Arduino\libraries -fqbn=esp8266:esp8266:generic:CpuFrequency=80,FlashFreq=40,FlashMode=dio,UploadSpeed=115200,FlashSize=4M1M,ResetMethod=nodemcu,Debug=Disabled,DebugLevel=None____ -ide-version=10611 -build-path C:\Users\User\AppData\Local\Temp\buildb8f9fbd0161026bef8078c61b5564ac2.tmp -warnings=all -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool.path=C:\Users\User\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.9 -prefs=runtime.tools.mkspiffs.path=C:\Users\User\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\0.1.2 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\User\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -verbose C:\Users\User\Documents\Arduino\libraries\easyMesh\examples\demoToy\demoToy.ino
Using board 'generic' from platform in folder: C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0
Using core 'esp8266' from platform in folder: C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0
Detecting libraries used...
"C:\Users\User\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include" "-IC:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include" "-IC:\Users\User\AppData\Local\Temp\buildb8f9fbd0161026bef8078c61b5564ac2.tmp/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10611 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_ESP01"  -DESP8266 "-IC:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266" "-IC:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\variants\generic" "C:\Users\User\AppData\Local\Temp\buildb8f9fbd0161026bef8078c61b5564ac2.tmp\sketch\demoToy.ino.cpp" -o "nul"
"C:\Users\User\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include" "-IC:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include" "-IC:\Users\User\AppData\Local\Temp\buildb8f9fbd0161026bef8078c61b5564ac2.tmp/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10611 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_ESP01"  -DESP8266 "-IC:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266" "-IC:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\variants\generic" "-IC:\Users\User\Documents\Arduino\libraries\ArduinoJson" "C:\Users\User\AppData\Local\Temp\buildb8f9fbd0161026bef8078c61b5564ac2.tmp\sketch\demoToy.ino.cpp" -o "nul"
"C:\Users\User\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include" "-IC:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include" "-IC:\Users\User\AppData\Local\Temp\buildb8f9fbd0161026bef8078c61b5564ac2.tmp/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10611 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_ESP01"  -DESP8266 "-IC:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266" "-IC:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\variants\generic" "-IC:\Users\User\Documents\Arduino\libraries\ArduinoJson" "-IC:\Users\User\Documents\Arduino\libraries\easyMesh\src" "C:\Users\User\AppData\Local\Temp\buildb8f9fbd0161026bef8078c61b5564ac2.tmp\sketch\demoToy.ino.cpp" -o "nul"
"C:\Users\User\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include" "-IC:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include" "-IC:\Users\User\AppData\Local\Temp\buildb8f9fbd0161026bef8078c61b5564ac2.tmp/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10611 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_ESP01"  -DESP8266 "-IC:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266" "-IC:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\variants\generic" "-IC:\Users\User\Documents\Arduino\libraries\ArduinoJson" "-IC:\Users\User\Documents\Arduino\libraries\easyMesh\src" "-IC:\Users\User\Documents\Arduino\libraries\SimpleList" "C:\Users\User\AppData\Local\Temp\buildb8f9fbd0161026bef8078c61b5564ac2.tmp\sketch\demoToy.ino.cpp" -o "nul"
"C:\Users\User\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include" "-IC:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include" "-IC:\Users\User\AppData\Local\Temp\buildb8f9fbd0161026bef8078c61b5564ac2.tmp/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fn [... cut ...]
Using library easyWebServer-master at version 1.0.0 in folder: C:\Users\User\Documents\Arduino\libraries\easyWebServer-master 
Using library easyWebSocket-master at version 1.0.0 in folder: C:\Users\User\Documents\Arduino\libraries\easyWebSocket-master 
Using library NeoPixelBus_by_Makuna at version 2.2.3 in folder: C:\Users\User\Documents\Arduino\libraries\NeoPixelBus_by_Makuna 
Using library SPI at version 1.0 in folder: C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\SPI 
exit status 1
Error compiling for board Generic ESP8266 Module.

sfranzyshen, 2016-10-27:
are you sure you are using Arduino 1.6.8 ??? I can see you are using 1.6.11 ... I have had problems with this version ... try 1.6.8 and it should compile ...

hubshi, 2016-10-27:
At last it worked, Phew, thanks for all of friends , it compiles now, i was using the SimpleList form the release folder , rar file,
from: https://github.com/blackhack/ArduLibraries/tree/master/Releases
while the working one is from : https://github.com/blackhack/ArduLibraries/tree/master/SimpleList
now after so many install/uninstall cycles , its working, you may add the working libs in your own repo, Thanks a lot for all of your help and support.

closed 2016-10-27 by hubshi