Coopdis/easyMesh #15: Can not Compile startHere Example

hubshi, 2016-10-26:
I am using ESP8266 ESP-12 , Arduino Version 1.6.11, ESP8266 Board Package Version 2.3.0. Following is the Error Log.

Arduino: 1.6.11 (Windows 7), Board: "Generic ESP8266 Module, 80 MHz, 40MHz, DIO, 115200, 4M (1M SPIFFS), nodemcu, Disabled, None"

C:\Development\Embedded\Arduino\Software\arduino-1.6.11\arduino-builder -dump-prefs -logger=machine -hardware C:\Development\Embedded\Arduino\Software\arduino-1.6.11\hardware -hardware C:\Users\User\AppData\Local\Arduino15\packages -hardware C:\Users\User\Documents\Arduino\hardware -tools C:\Development\Embedded\Arduino\Software\arduino-1.6.11\tools-builder -tools C:\Development\Embedded\Arduino\Software\arduino-1.6.11\hardware\tools\avr -tools C:\Users\User\AppData\Local\Arduino15\packages -built-in-libraries C:\Development\Embedded\Arduino\Software\arduino-1.6.11\libraries -libraries C:\Users\User\Documents\Arduino\libraries -fqbn=esp8266:esp8266:generic:CpuFrequency=80,FlashFreq=40,FlashMode=dio,UploadSpeed=115200,FlashSize=4M1M,ResetMethod=nodemcu,Debug=Disabled,DebugLevel=None____ -ide-version=10611 -build-path C:\Users\User\AppData\Local\Temp\buildedca15703cc72be4b2ec34340762f789.tmp -warnings=all -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool.path=C:\Users\User\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.9 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\User\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -prefs=runtime.tools.mkspiffs.path=C:\Users\User\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\0.1.2 -verbose C:\Users\User\Documents\Arduino\libraries\easyMesh\examples\startHere\startHere.ino
C:\Development\Embedded\Arduino\Software\arduino-1.6.11\arduino-builder -compile -logger=machine -hardware C:\Development\Embedded\Arduino\Software\arduino-1.6.11\hardware -hardware C:\Users\User\AppData\Local\Arduino15\packages -hardware C:\Users\User\Documents\Arduino\hardware -tools C:\Development\Embedded\Arduino\Software\arduino-1.6.11\tools-builder -tools C:\Development\Embedded\Arduino\Software\arduino-1.6.11\hardware\tools\avr -tools C:\Users\User\AppData\Local\Arduino15\packages -built-in-libraries C:\Development\Embedded\Arduino\Software\arduino-1.6.11\libraries -libraries C:\Users\User\Documents\Arduino\libraries -fqbn=esp8266:esp8266:generic:CpuFrequency=80,FlashFreq=40,FlashMode=dio,UploadSpeed=115200,FlashSize=4M1M,ResetMethod=nodemcu,Debug=Disabled,DebugLevel=None____ -ide-version=10611 -build-path C:\Users\User\AppData\Local\Temp\buildedca15703cc72be4b2ec34340762f789.tmp -warnings=all -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool.path=C:\Users\User\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.9 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\User\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -prefs=runtime.tools.mkspiffs.path=C:\Users\User\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\0.1.2 -verbose C:\Users\User\Documents\Arduino\libraries\easyMesh\examples\startHere\startHere.ino
Using board 'generic' from platform in folder: C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0
Using core 'esp8266' from platform in folder: C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0
Detecting libraries used...
"C:\Users\User\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include" "-IC:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include" "-IC:\Users\User\AppData\Local\Temp\buildedca15703cc72be4b2ec34340762f789.tmp/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10611 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_ESP01"  -DESP8266 "-IC:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266" "-IC:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\variants\generic" "C:\Users\User\AppData\Local\Temp\buildedca15703cc72be4b2ec34340762f789.tmp\sketch\startHere.ino.cpp" -o "nul"
"C:\Users\User\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include" "-IC:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include" "-IC:\Users\User\AppData\Local\Temp\buildedca15703cc72be4b2ec34340762f789.tmp/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10611 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_ESP01"  -DESP8266 "-IC:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266" "-IC:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\variants\generic" "-IC:\Users\User\Documents\Arduino\libraries\easyMesh\src" "C:\Users\User\AppData\Local\Temp\buildedca15703cc72be4b2ec34340762f789.tmp\sketch\startHere.ino.cpp" -o "nul"
"C:\Users\User\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include" "-IC:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include" "-IC:\Users\User\AppData\Local\Temp\buildedca15703cc72be4b2ec34340762f789.tmp/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10611 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_ESP01"  -DESP8266 "-IC:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266" "-IC:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\variants\generic" "-IC:\Users\User\Documents\Arduino\libraries\easyMesh\src" "-IC:\Users\User\Documents\Arduino\libraries\SimpleList" "C:\Users\User\AppData\Local\Temp\buildedca15703cc72be4b2ec34340762f789.tmp\sketch\startHere.ino.cpp" -o "nul"
"C:\Users\User\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include" "-IC:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include" "-IC:\Users\User\AppData\Local\Temp\buildedca15703cc72be4b2ec34340762f789.tmp/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10611 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_ESP01"  -DESP8266 "-IC:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266" "-IC:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\variants\generic" "-IC:\Users\User\Documents\Arduino\libraries\easyMesh\src" "-IC:\Users\User\Documents\Arduino\libraries\SimpleList" "-IC:\Users\User\Documents\Arduino\libraries\ArduinoJson" "C:\Users\User\AppData\Local\Temp\buildedca15703cc72be4b2ec34340762f789.tmp\sketch\startHere.ino.cpp" -o "nul"
"C:\Users\User\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include" "-IC:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include" "-IC:\Users\User\AppData\Local\Temp\buildedca15703cc72be4b2ec34340762f789.tmp/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fn [... cut ...]
C:\Users\User\Documents\Arduino\libraries\SimpleList/SimpleList.h:144:33: note: synthesized method 'meshConnectionType& meshConnectionType::operator=(const meshConnectionType&)' first required here 

                     newArray[i] = _internalArray[i + 1];

                                 ^

Using library easyMesh at version 1.0.0 in folder: C:\Users\User\Documents\Arduino\libraries\easyMesh 
Using library SimpleList in folder: C:\Users\User\Documents\Arduino\libraries\SimpleList (legacy)
Using library ArduinoJson at version 5.6.7 in folder: C:\Users\User\Documents\Arduino\libraries\ArduinoJson 
exit status 1
Error compiling for board Generic ESP8266 Module.

RudyFiero, 2016-10-26:
I had similar problems before. I ended up deleting the library and the two external libraries, redownload and reinstall. Then the errors disappeared.

muratdemirtas, 2016-10-26:
nice story bro, please update your arduino core.

Installing with Boards Manager

Starting with 1.6.4, Arduino allows installation of third-party platform packages using Boards Manager. We have packages available for Windows, Mac OS, and Linux (32 and 64 bit).

Install Arduino 1.6.8 from the Arduino website.
Start Arduino and open Preferences window.
Enter http://arduino.esp8266.com/stable/package_esp8266com_index.json into Additional Board Manager URLs field. You can add multiple URLs, separating them with commas.
Open Boards Manager from Tools > Board menu and install esp8266 platform (and don't forget to select your ESP8266 board from Tools > Board menu after installation).

hubshi, 2016-10-26:
Thanks for quick reply,
I have followed your suggestions, but again the output error is same, Kindly check the following error log:

C:\Users\User\Documents\Arduino\libraries\easyMesh\src\eashMeshConnection.cpp: In member function 'void easyMesh::manageConnections()':

C:\Users\User\Documents\Arduino\libraries\easyMesh\src\eashMeshConnection.cpp:65:16: warning: enumeration value 'REQUESTED' not handled in switch [-Wswitch]

         switch ( connection->nodeSyncStatus ) {

                ^

C:\Users\User\Documents\Arduino\libraries\easyMesh\src\eashMeshConnection.cpp:65:16: warning: enumeration value 'COMPLETE' not handled in switch [-Wswitch]

C:\Users\User\Documents\Arduino\libraries\easyMesh\src\eashMeshConnection.cpp:76:16: warning: enumeration value 'REQUESTED' not handled in switch [-Wswitch]

         switch ( connection->timeSyncStatus ) {

                ^

C:\Users\User\Documents\Arduino\libraries\easyMesh\src\eashMeshConnection.cpp:76:16: warning: enumeration value 'COMPLETE' not handled in switch [-Wswitch]

C:\Users\User\Documents\Arduino\libraries\easyMesh\src\eashMeshConnection.cpp: In member function 'meshConnectionType* easyMesh::findConnection(espconn*)':

C:\Users\User\Documents\Arduino\libraries\easyMesh\src\eashMeshConnection.cpp:142:9: warning: unused variable 'i' [-Wunused-variable]

     int i=0;

         ^

C:\Users\User\Documents\Arduino\libraries\easyMesh\src\eashMeshConnection.cpp: At global scope:

C:\Users\User\Documents\Arduino\libraries\easyMesh\src\eashMeshConnection.cpp:273:24: warning: unused parameter 'length' [-Wunused-parameter]

 void ICACHE_FLASH_ATTR easyMesh::meshRecvCb(void *arg, char *data, unsigned short length) {

                        ^

C:\Users\User\Documents\Arduino\libraries\easyMesh\src\eashMeshConnection.cpp:377:24: warning: unused parameter 'arg' [-Wunused-parameter]

 void ICACHE_FLASH_ATTR easyMesh::meshReconCb(void *arg, sint8 err) {

                        ^

In file included from C:\Users\User\Documents\Arduino\libraries\easyMesh\src\eashMeshConnection.cpp:18:0:

C:\Users\User\Documents\Arduino\libraries\easyMesh\src\easyMesh.h: In instantiation of 'T* SimpleList<T>::erase(SimpleList<T>::iterator) [with T = meshConnectionType; SimpleList<T>::iterator = meshConnectionType*]':

C:\Users\User\Documents\Arduino\libraries\easyMesh\src\eashMeshConnection.cpp:44:37:   required from here

C:\Users\User\Documents\Arduino\libraries\easyMesh\src\easyMesh.h:69:8: error: no matching function for call to 'SimpleList<String>::SimpleList(const SimpleList<String>&)'

 struct meshConnectionType {

        ^

C:\Users\User\Documents\Arduino\libraries\easyMesh\src\easyMesh.h:69:8: note: candidates are:

In file included from C:\Users\User\Documents\Arduino\libraries\easyMesh\src\eashMeshConnection.cpp:11:0:

C:\Users\User\Documents\Arduino\libraries\SimpleList/SimpleList.h:39:5: note: SimpleList<T>::SimpleList(SimpleList<T>&) [with T = String]

     SimpleList(SimpleList& from)

     ^

C:\Users\User\Documents\Arduino\libraries\SimpleList/SimpleList.h:39:5: note:   no known conversion for argument 1 from 'const SimpleList<String>' to 'SimpleList<String>&'

C:\Users\User\Documents\Arduino\libraries\SimpleList/SimpleList.h:22:5: note: SimpleList<T>::SimpleList() [with T = String]

     SimpleList()

     ^

C:\Users\User\Documents\Arduino\libraries\SimpleList/SimpleList.h:22:5: note:   candidate expects 0 arguments, 1 provided

C:\Users\User\Documents\Arduino\libraries\SimpleList/SimpleList.h:51:17: error:   initializing argument 1 of 'SimpleList<T>& SimpleList<T>::operator=(SimpleList<T>) [with T = String]'

     SimpleList& operator=(SimpleList from)

                 ^

C:\Users\User\Documents\Arduino\libraries\SimpleList/SimpleList.h:144:33: note: synthesized method 'meshConnectionType& meshConnectionType::operator=(const meshConnectionType&)' first required here 

                     newArray[i] = _internalArray[i + 1];

                                 ^

Using library easyMesh at version 1.0.0 in folder: C:\Users\User\Documents\Arduino\libraries\easyMesh 
Using library SimpleList in folder: C:\Users\User\Documents\Arduino\libraries\SimpleList (legacy)
Using library ArduinoJson at version 5.6.7 in folder: C:\Users\User\Documents\Arduino\libraries\ArduinoJson 
exit status 1
Error compiling for board WeMos D1(Retired).

muratdemirtas, 2016-10-26:
pls reinstall simplelist and arduinojson libraries.

muratdemirtas, 2016-10-26:
Error compiling for board WeMos D1(Retired). Pls select nodemcu 1.0 esp-12e for compiling

RudyFiero, 2016-10-26:
Board: "Generic ESP8266 Module" works for me. And as I have said above, I had what looks to be similar errors. It was frustrating because I deleted all the libraries and then installed new ones, and it still didn't work. But I did it all again and then it has been working since. I never thought of it at the time but I did end up re-starting my machine in the end. 

I don't know what this happened but I don't believe it has anything to do with a problem with this library other than requiring the current versions of the libraries specified.

hubshi, 2016-10-27:
I have tried it in a new VM now, everything fresh installed , same errors, as you suggested , i downloaded the libs from their respective Git respos, pulling my hair. can you please share your libs and arduino installation by making a rar file so i can try it , mainly i think this is a problem of compiler because the error is coming from the template, of simplelist.h , as the author also explained that he made all of the functions in .h file because compiler do not like templates in .cpp file.

> // All implementation in .h because avr goes crazy with templates in .cpp

BTW examples of simplelist compiles well, but all examples use list of int, while easymesh uses list of string,

RudyFiero, 2016-10-27:
I don't have anything helpful to add. But I had the same problems that you are having. I had reloaded everything a couple of times. Then one day I tried again, and it failed. Then I did it again and it worked. I don't know what the problem was or the sequence I finally used. I know how frustrating this is.

sfranzyshen, 2016-10-27:
are you sure you are using Arduino 1.6.8 ???

hubshi, 2016-10-27:
Once again i started with clean install, on a new VM, same problem, following is the error log. and yes i am using only one instance of simplelist, with Arduino IDE 1.6.8 and ESP8266 2.3.0, and whatever i select the model of ESP8266, same error.

C:\Users\User\Documents\Arduino\libraries\easyMesh\src\eashMeshConnection.cpp: In member function 'void easyMesh::manageConnections()':

C:\Users\User\Documents\Arduino\libraries\easyMesh\src\eashMeshConnection.cpp:65:16: warning: enumeration value 'REQUESTED' not handled in switch [-Wswitch]

         switch ( connection->nodeSyncStatus ) {

                ^

C:\Users\User\Documents\Arduino\libraries\easyMesh\src\eashMeshConnection.cpp:65:16: warning: enumeration value 'COMPLETE' not handled in switch [-Wswitch]

C:\Users\User\Documents\Arduino\libraries\easyMesh\src\eashMeshConnection.cpp:76:16: warning: enumeration value 'REQUESTED' not handled in switch [-Wswitch]

         switch ( connection->timeSyncStatus ) {

                ^

C:\Users\User\Documents\Arduino\libraries\easyMesh\src\eashMeshConnection.cpp:76:16: warning: enumeration value 'COMPLETE' not handled in switch [-Wswitch]

C:\Users\User\Documents\Arduino\libraries\easyMesh\src\eashMeshConnection.cpp: In member function 'meshConnectionType* easyMesh::findConnection(espconn*)':

C:\Users\User\Documents\Arduino\libraries\easyMesh\src\eashMeshConnection.cpp:142:9: warning: unused variable 'i' [-Wunused-variable]

     int i=0;

         ^

C:\Users\User\Documents\Arduino\libraries\easyMesh\src\eashMeshConnection.cpp: At global scope:

C:\Users\User\Documents\Arduino\libraries\easyMesh\src\eashMeshConnection.cpp:273:24: warning: unused parameter 'length' [-Wunused-parameter]

 void ICACHE_FLASH_ATTR easyMesh::meshRecvCb(void *arg, char *data, unsigned short length) {

                        ^

C:\Users\User\Documents\Arduino\libraries\easyMesh\src\eashMeshConnection.cpp:377:24: warning: unused parameter 'arg' [-Wunused-parameter]

 void ICACHE_FLASH_ATTR easyMesh::meshReconCb(void *arg, sint8 err) {

                        ^

In file included from C:\Users\User\Documents\Arduino\libraries\easyMesh\src\eashMeshConnection.cpp:18:0:

C:\Users\User\Documents\Arduino\libraries\easyMesh\src\easyMesh.h: In instantiation of 'T* SimpleList<T>::erase(SimpleList<T>::iterator) [with T = meshConnectionType; SimpleList<T>::iterator = meshConnectionType*]':

C:\Users\User\Documents\Arduino\libraries\easyMesh\src\eashMeshConnection.cpp:44:37:   required from here

C:\Users\User\Documents\Arduino\libraries\easyMesh\src\easyMesh.h:69:8: error: no matching function for call to 'SimpleList<String>::SimpleList(const SimpleList<String>&)'

 struct meshConnectionType {

        ^

C:\Users\User\Documents\Arduino\libraries\easyMesh\src\easyMesh.h:69:8: note: candidates are:

In file included from C:\Users\User\Documents\Arduino\libraries\easyMesh\src\eashMeshConnection.cpp:11:0:

C:\Users\User\Documents\Arduino\libraries\SimpleList/SimpleList.h:39:5: note: SimpleList<T>::SimpleList(SimpleList<T>&) [with T = String]

     SimpleList(SimpleList& from)

     ^

C:\Users\User\Documents\Arduino\libraries\SimpleList/SimpleList.h:39:5: note:   no known conversion for argument 1 from 'const SimpleList<String>' to 'SimpleList<String>&'

C:\Users\User\Documents\Arduino\libraries\SimpleList/SimpleList.h:22:5: note: SimpleList<T>::SimpleList() [with T = String]

     SimpleList()

     ^

C:\Users\User\Documents\Arduino\libraries\SimpleList/SimpleList.h:22:5: note:   candidate expects 0 arguments, 1 provided

C:\Users\User\Documents\Arduino\libraries\SimpleList/SimpleList.h:51:17: error:   initializing argument 1 of 'SimpleList<T>& SimpleList<T>::operator=(SimpleList<T>) [with T = String]'

     SimpleList& operator=(SimpleList from)

                 ^

C:\Users\User\Documents\Arduino\libraries\SimpleList/SimpleList.h:144:33: note: synthesized method 'meshConnectionType& meshConnectionType::operator=(const meshConnectionType&)' first required here 

                     newArray[i] = _internalArray[i + 1];

                                 ^

Using library easyMesh at version 1.0.0 in folder: C:\Users\User\Documents\Arduino\libraries\easyMesh 
Using library SimpleList in folder: C:\Users\User\Documents\Arduino\libraries\SimpleList (legacy)
Using library ArduinoJson at version 5.6.7 in folder: C:\Users\User\Documents\Arduino\libraries\ArduinoJson 
exit status 1
Error compiling for board NodeMCU 0.9 (ESP-12 Module).

sfranzyshen, 2016-10-27:
make sure you are using https://github.com/blackhack/ArduLibraries/tree/master/SimpleList
and not something else ...

hubshi, 2016-10-27:
At last it worked, Phew, thanks for all of friends , it compiles now, i was using the SimpleList form the release folder , rar file, from: https://github.com/blackhack/ArduLibraries/tree/master/Releases
but now after so many install/uninstall cycles , its working, you may add the working libs in your own repo, Thanks a lot for all of your help and support.

closed 2016-10-27 by hubshi