Coopdis/easyMesh #16: cannot compile start here example

jasonmarkham, 2016-10-27:
while compiling i am getting this error 
i tried to change module selection but still cannot compile it . ! any one had this error ? 

C:\Users\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0/tools/sdk/include/ip_addr.h:56:128: error: invalid operands of types 'ip_addr*' and 'int' to binary 'operator>>'

 #define IP2STR(addr) (uint8_t)(addr & 0xFF), (uint8_t)((addr >> 8) & 0xFF), (uint8_t)((addr >> 16) & 0xFF), (uint8_t)((addr >> 24) & 0xFF)

                                                                                                                                ^

C:\Users\Documents\Arduino\libraries\easyMesh-master\src\easyMeshAP.cpp:40:19: note: in expansion of macro 'IP2STR'

                   IP2STR( &ipInfo.netmask ) );

                   ^

exit status 1
Error compiling for board NodeMCU 0.9 (ESP-12 Module).

hubshi, 2016-10-27:
1. Remove your esp8266 package from the Boards Manager,
2. Close Arduino IDE
3. Delete this folder :  %AppData%\Local\Arduino15\packages\esp8266\
4. Delete this folder :  %AppData%\Local\Arduino15\staging\packages
5. Open Arduino IDE
6. Install Esp8266 package from Boards Manager

I hope it will work

RudyFiero, 2016-10-27:
Or if you want the long version  https://github.com/Coopdis/easyMesh/issues/15

jasonmarkham, 2016-10-28:
Thanks for your guide , it worked fine now !

Kudos to you !

Tattvamaso, 2017-01-24:
Hi all, this is my first post on Git so have mercy:)
I cannot compile "starthere" because of a lot of undefined stuff e.g. meshSentCb. I thought I knew why: there is a typo in eashMeshConnection.cpp, But changing that into easy (y instead of h ) didn solve the issue.
Running Ubuntu 16.04 64 bit and Arduino 1.6.13. All other suggestions as only one lib and correct SimpleList.h are done. It seems like there is missing a referral but I cannot find to what. PArt of the debug:
 easyMeshSTA.cpp:29: undefined reference to `easyMesh::meshSentCb(void*)'
~/Arduino/libraries/easyMesh/easyMeshSTA.cpp:29: undefined reference to `easyMesh::meshReconCb(void*, signed char)'
libraries/easyMesh/easyMeshSTA.cpp.o: In function `easyMesh::scanTimerCallback(void*)':
~Arduino/libraries/easyMesh/easyMeshSTA.cpp:87: undefined reference to `easyMesh::meshDisconCb(void*)'

What am I missing?
Rob