Coopdis/easyMesh #19: question: Get Telemetry from each Mesh Node

jaecktec, 2016-12-08:
I haven't seen a way to get telemtry data (like signal strength and nodes in proximidity). Is there a way?

BlackEdder, 2016-12-22:
There is no build in way to do this, but this can be done in your own code by scanning for APs.

An example that might get you started:
https://gitlab.com/BlackEdder/painlessMesh/blob/master/src/painlessMeshSTA.cpp#L124

By the way. I am not sure how active this project is anymore, but am maintaining a fork here:
https://gitlab.com/BlackEdder/painlessMesh
That also incorporates @sfranzyshen changes.

RudyFiero, 2016-12-22:
I'm going to try this as soon as I can.

jaecktec, 2016-12-23:
Thank you! The AP way is quite unrelaible. (Antenna diversity and stuff,..) But thank you :)
Hope I can get the Raw Packets library set working and build a time of flight thingy...

closed 2017-01-30 by jaecktec