Coopdis/easyMesh #20: Node time sync is not achieved

gmag11, 2016-12-18:
I'm trying to do a simple setup with 2 esp8266 to. My intention is to have its LED blinking synchronously, so time sync is needed. I've used StatHere example.

I guess there is a bug while calculating adjustment value but I've been not able to find it.

My code is this:
```
//************************************************************
// this is a simple example that uses the easyMesh library
//
// 1. blinks led once for every node on the mesh
// 2. blink cycle repeats every BLINK_PERIOD
// 3. sends a silly message to every node on the mesh at a random time betweew 1 and 5 seconds
// 4. prints anything it recieves to Serial.print
// 
//
//************************************************************
#include <easyMesh.h>

// some gpio pin that is connected to an LED... 
// on my rig, this is 5, change to the right number of your LED.
#define   LED             2       // GPIO number of connected LED

#define   BLINK_PERIOD    1000000 // microseconds until cycle repeat
#define   BLINK_DURATION  100000  // microseconds LED is on for

#define   MESH_PREFIX     "whateverYouLike"
#define   MESH_PASSWORD   "somethingSneeky"
#define   MESH_PORT       5555

easyMesh  mesh;

uint32_t sendMessageTime = 0;

void setup() {
  Serial.begin(115200);
    
  pinMode( LED, OUTPUT );

//mesh.setDebugMsgTypes( ERROR | MESH_STATUS | CONNECTION | SYNC | COMMUNICATION | GENERAL | MSG_TYPES | REMOTE ); // all types on
  mesh.setDebugMsgTypes( ERROR | STARTUP | SYNC );  // set before init() so that you can see startup messages

  mesh.init( MESH_PREFIX, MESH_PASSWORD, MESH_PORT );
  mesh.setReceiveCallback( &receivedCallback );
  mesh.setNewConnectionCallback( &newConnectionCallback );

  randomSeed( analogRead( A0 ) );  
}

void loop() {
  mesh.update();

  // run the blinky
  bool  onFlag = true;
  uint32_t cycleTime = mesh.getNodeTime() % BLINK_PERIOD;
  for ( uint8_t i = 0; i < ( mesh.connectionCount() + 1); i++ ) {
    uint32_t onTime = BLINK_DURATION * i * 2;    

    if ( cycleTime > onTime && cycleTime < onTime + BLINK_DURATION )
      onFlag = false;
  }
  digitalWrite( LED, onFlag );

  // get next random time for send message
  if ( sendMessageTime == 0 ) {
    sendMessageTime = mesh.getNodeTime() + random( 1000000, 5000000 );
  }

  // if the time is ripe, send everyone a message!
  if ( sendMessageTime != 0 && sendMessageTime < mesh.getNodeTime() ){
    String msg = "Hello from node ";
[... 319 lines cut ...]
0x10	buildTimeStamp(): num=9
0x10	buildTimeStamp(): timeStamp={"time":18227919,"num":9,"adopt":true}
0x10	handleTimeSync(): with 12617095 out timestamp={"time":18227919,"num":9,"adopt":true}
startHere: New Connection, adopt=1
0x10	handleTimeSync(): with 12617095 in timestamp={"time":19606042,"num":9,"adopt":true}
0x10	processTimeStamp(): str={"time":19606042,"num":9,"adopt":true}
0x10	handleTimeSync(): with 12617095 out timestamp={"time":19606042,"num":9,"adopt":true}
0x10	calcAdjustment(): odd=0
0x10	best interval=67552, best index=0
0x10	new calc time=19240242, adoptedTime=19240242
0x10	manageConnections(): start nodeSync with 12617095
0x10	startNodeSync(): with 12617095
0x10	handleNodeSync(): with 12617095
0x10	handleNodeSync(): valid NODE_SYNC_REPLY from 12617095
startHere: Received from 12617095 msg=Hello from node 12617095
Node time: 21951040
0x10	handleNodeSync(): with 12617095
0x10	handleNodeSync(): valid NODE_SYNC_REQUEST 12617095 sending NODE_SYNC_REPLY
0x10	handleNodeSync(): with 12617095
0x10	handleNodeSync(): valid NODE_SYNC_REQUEST 12617095 sending NODE_SYNC_REPLY
startHere: Received from 12617095 msg=Hello from node 12617095
Node time: 25705848
0x10	handleNodeSync(): with 12617095
0x10	handleNodeSync(): valid NODE_SYNC_REQUEST 12617095 sending NODE_SYNC_REPLY
```

ds1982, 2017-07-21:
I have exactly the same behaviour and NodeTimes are not in sync while that. 
I am also using this with a slowly blinking LED and the Notetimes differ ~1 second.

Anyone has a solution?

sfranzyshen, 2017-07-23:
this code is dead ... try painlessmesh