Coopdis/easyMesh #21: How to connect to a wifi network with internet connection from any node?

sammj007, 2017-01-23:
I would like to connect to internet from any of the nodes in the mesh, and push some information based on the messages received by the node. Is there a way to do this?
I could not figure out if the nodes connect to each other (when there are just two nodes), or only one of them connect to the node acting as an AP.
Any suggestions on how to achieve internet connectivity from a mesh?

BlackEdder, 2017-01-24:
We discussed this possibility in our fork of easyMesh:
https://gitlab.com/BlackEdder/painlessMesh/issues/26 (last two comments), but due to the design of the network it is not straightforward. 

I do maintain a simple project [1] that I use to connect to the Mesh from a computer, but the computer still needs to connect to the mesh itself. In practice you need two wifi cards (one for normal network, one for mesh) or you need to switch between the two networks.

[1] https://github.com/BlackEdder/easyMesheD

sammj007, 2017-01-24:
Thanks, will check the fork.
BTW, I don't have an option of using a computer, since I need this mesh to be only using ESP8266.

BlackEdder, 2017-01-24:
Then (AFAIK) your only option would be to disconnect from the mesh, connect
to the other network, send the messages and then reconnect to the mesh.

sammj007, 2017-01-24:
So, even the painlessMesh also doesn't have this support? I was thinking that since ESP supports dual mode (AP + Station), any one node which runs as AP can connect to the network with internet, and stay connected? Maybe I didn't fully follow how the mesh works, hence such a thought!

BlackEdder, 2017-01-24:
We have a description on the way the mesh (easyMesh/painlessMesh) is setup
here:
https://gitlab.com/BlackEdder/painlessMesh/wikis/home
Based on the mesh layout described there I think having an AP node might be
problematic, because it is not guaranteed that the mesh will connect to it.

Now it should be relatively easy to implement an AP only HYBRID mode (see
the PR I linked above, where we implemented a STA only mode). If you are
happy to experiment and find that it (mostly) works anyway I'd happily
accept a PR on the issue.

sammj007, 2017-01-24:
Sure, will give a try.
Thanks for your inputs.

BlackEdder, 2017-01-24:
Thinking about it a bit more I don't think an AP only mode will be much of
a problem.  Feel free to open issue at painlessMesh.  Of course the fastest
way to get it implemented would be to open a PR.

sammj007, 2017-01-25:
Will have my further questions post at painlesMesh :)