Coopdis/easyMesh #25: mesh.sendSingle() causes: "0x1	meshRecvCb: parseObject() failed. data=dest<--" error

sglvladi, 2017-02-28:
Hi there,

I discovered your library today and thought I'd do some experimenting in order to get the mesh network connected to WiFi/MQTT by introducing a gateway node composed of two ESP8266s. One ESP is connected to our home MQTT server, while the other is connected to the mesh and the two talk to each other through Software Serial. After a bit of fiddling around, I have managed to get this part all running smoothly. (Happy days!!)

Overall I have 3 nodes in my disposal, which I am using for testing and are listed below:

1. **Gateway Node:** Dual ESP8266 - as described above (Only one is connected to the mesh)
2. **LED Node**: Single ESP8266 - Has a LED, which is controlled by external MQTT server.
3. **Dummy Node**: Single ESP8266 - Does pretty much nothing. Simply there to test mesh functionality.

So, the test case scenario involves the evaluation of the communication between the Gateway and LED nodes. The Gateway will forward the LED state to the LED node and then the latter will respond to confirm that the state has been changed. 

The code used to send the message from the Gateway to the LED node is as follows:
```c++
    uint32_t dest = 12844338;
    mesh.sendSingle(dest, swMessage); 
    // swMessage is a String read from the Software Serial 
```
Everything works fine when the nodes are connected directly . The problem occurs when the Dummy node adopts either the Gateway or LED nodes.  When this happens, the following error gets displayed on the LED node (as quoted in title):

```c++
0x1	meshRecvCb: parseObject() failed. data=dest<--
```

I should also note, that the problem does not occur in the case where `mesh.sendSingle(dest, swMessage); ` is replaced by `mesh.sendBroadcast(swMessage);`

Any help to resolve the issue would be much appreciated. :)

Many thanks in advance.

RudyFiero, 2017-02-28:
@sglvladi 
Coopdis seems to have disappeared last fall and there has been no work or updates done. I don't expect you will get a reply from Coopdis on the issue you raised. If you find that easyMesh does not do what you need then maybe you should look at painlessMesh that was based on easyMesh but has people working on it.   https://gitlab.com/BlackEdder/painlessMesh

sglvladi, 2017-02-28:
@RudyFiero 
Thanks very much for the quick response! I'll have a go at painlessMesh then.

sglvladi, 2017-03-01:
Wow! painlessMesh works like a Charm! Just to save the effort for anyone else, DO NOT BOTHER WITH easyMesh, JUMP onto painlessMesh!! Kudos to the guys that maintain it :D

closed 2017-03-01 by sglvladi