Coopdis/easyMesh #28: error: 'next' was not declared in this scope

Defozo, 2017-03-25:
```
sketch\easyMeshSTA.cpp: In static member function 'static void easyMesh::stationScanCb(void*, STATUS)':

sketch\easyMeshSTA.cpp:107:40: error: 'next' was not declared in this scope

         bssInfo = STAILQ_NEXT(bssInfo, next);

                                        ^

sketch\easyMeshSTA.cpp:107:44: error: 'STAILQ_NEXT' was not declared in this scope

         bssInfo = STAILQ_NEXT(bssInfo, next);

                                            ^

exit status 1
Error compiling.
```

danidask, 2017-04-23:
Same thing is happening to me

BlackEdder, 2017-04-24:
As far as I can tell easyMesh is abandoned. I am maintaining a fork on gitlab though:
https://gitlab.com/BlackEdder/painlessMesh