Coopdis/easyMesh #30: Raspberry Pi as mesh node

nemo9955, 2017-03-31:
I want to write a python port for this library that basically allows a Raspberry/Orange/Banana/Panda board to create a wifi hotspot and have at least one ESP8266 node connected to it.

After some easyMesh source code skimming, I'm thinking of doing something like this : 
- create a wifi hotspot with a name like _meshPrefix+"00001" and _meshPassword that is started before all other nodes
- a python script (flask maybe ?) that listens on _meshPort and parses the packet/message
- the python script is also able to broadcast a message to the mesh

What do I want from this port, ordered from most important to least :
- the raspberry is able to receive/listen to every broadcast witch will contain some sensors readings that I will put in a database (influxDB)
- have a updated list of all the nodes and their connections to display a visual representation of the graph
- send a message to individual/every node, the esp will parse that message into a command (IR emitter, LED state, OLED display message )

BlackEdder, 2017-04-03:
I wrote a [easyMesheD](https://github.com/BlackEdder/easyMesheD) that can receive messages and send messages to the mesh. It does need to actively connect to the mesh, so if you want to also connect to the wider network at the same time then you need two network cards on your raspberry pi.

I am currently writing an improved version as well that basically act as a full node as part of [painlessMesh](https://gitlab.com/BlackEdder/painlessMesh/merge_requests/33), which is a fork of easyMesh (easyMesh seems to have been abandoned).

nemo9955, 2017-04-03:
This seems like a good enough solution, thank you!
I'll try to put it on a raspberry and check it out.

durvesh09, 2017-07-10:
Hi,

   Had you succeeded in using raspberry pi as a mesh node ?

nemo9955, 2017-07-10:
No, I had problems with the whole mesh and postponed it.
There is this working library, but again, some mesh problems and the fact that it's written in D  made me not bother with finding a solution.

https://gitlab.com/BlackEdder/painlessMeshListener

durvesh09, 2017-07-11:
Thanks. But the painlessMeshListener is not working. Infact there is no any procedure regarding its usage.

BlackEdder, 2017-07-11:
I put some notes on its usage on the painlessMesh wiki and in the readme (although indeed minimal). Maybe if you raise some issues I can help you solve your problem. You can also try to contact me on gitter. https://gitter.im/painlessMesh/Lobby