Coopdis/easyMesh #31: Connections and starting

Nickson2000, 2017-04-04:
I got Arduino Nano and ESP8266
Please say me: 
-what to connect 8266 to Arduino if I need to start EasyMesh Project
-it is nesessary to flash 8266 with some new firmware?
Thank you!

SupotSaeEa, 2017-04-28:
No need Arduino nano.
ESP8266 use this library itself.
It mean that you can compile and upload code to ESP8266.

ujjwalrathod007, 2017-06-16:
I want to make a Mesh using two ESP8266. I have some questions.
First I have to add the method in both ESP void easyMesh::init( String prefix, String password, uint16_t port ) And some others as mentioned on the main page, So it creates 2 Mesh with different parameters If I want to send data from one of them I use the following function. 

void receivedCallback( uint32_t from, String &msg )

But I am not sure that will the other node receive it? Can anyone tell my idea is correct?

SupotSaeEa, 2017-06-17:
@ujjwalrathod007 You should use the same parameter on both ESP.
void receivedCallback.. is event handler when something received.
When you need to send something out. You can use mesh.sendBroadcast(something);
All other node will receive it.

ujjwalrathod007, 2017-07-05:
Actually, I am making the network using two ESPs at present. One has sensor connected and the other one receives the data. The problem is the routine mesh.sendBroadcast(something) evaluates to TRUE but on the receiver side I don't see anything. On my serial monitor I see only blank stuffs which actually scrolls down. Do I receive anything? actually void receivedCallback cannot be checked weather it is TRUE or FALSE.