Coopdis/easyMesh #37: Connecting to Web

unishubh, 2017-08-30:
Let us assume we have 5 ESP connected as 1<-2<-3<-4<-5 .

Can i connect esp 1 to wifi router thus enabling internet connectivity to my mesh, and then make a request to some external website from node 2,3,4 or 5 ??
Can this be done? If yes the How?

josep112, 2017-09-01:
I am also interested in this condition

BlackEdder, 2017-09-09:
As far as I can tell easyMesh is abandoned. I am maintaining a fork on gitlab though:
https://gitlab.com/BlackEdder/painlessMesh

To answer your question. With easymesh this can't be done. In painlessMesh the first step can be done soon (will merge support in a couple of days). For the second step you will have to have node 1 send the relevant information to node 2,3,4,5