Coopdis/easyMesh #40: EasyMesh vs PainlessMesh

AlexisTM, 2017-10-15:
Hi there,

Who is the copy of whom? EasyMesh or PainlessMesh?

One is MIT, the other GPLv3, but it's almost the same codebase.

https://gitlab.com/BlackEdder/painlessMesh/blob/master/README.md

NicksonYap, 2017-10-22:
PainlessMesh's Repo description says:
"A painless way to setup a mesh with ESP8266 and ESP32 devices. Based on https://github.com/Coopdis/easyMesh"

danielcera, 2018-04-11:
I am working with PainlessMesh, but with more than 10 nodes it is something unstable, it will be that with easymesh I will achieve a better performance, or what is the difference between these two, thanks.

AlexisTM, 2018-04-11:
@danielcera I guess the best way to know is to test.

Plus, if you report this to PainlessMesh, you can fix the problem at the source :)

This repo is on sleep mode unlike painlessMesh ;)

julioj4pa, 2018-08-15:
@danielcera Is it unstable to send broadcast package and unicast package?

AlexisTM, 2024-10-28:
Outdated.

closed 2024-10-28 by AlexisTM