Coopdis/easyMesh #46: Cannot compile startHere example

Naff16, 2018-01-26:
Im having probems in compiling the startHere example. Im using Arduino IDE 1.8.3, already downloaded SimpleList, ArduinoJson. Updated esp8266 packages (2.4.0) and removed the old ones. Im getting this error when trying to compile using the  NodeMCU 1.0(ESP-12E Module).

`In file included from /home/andre/.arduino15/packages/esp8266/hardware/esp8266/2.4.0/tools/sdk/lwip2/include/lwip/ip_addr.h:43:0,
                 from /home/andre/.arduino15/packages/esp8266/hardware/esp8266/2.4.0/tools/sdk/include/user_interface.h:30,
                 from /home/andre/arduino-1.8.3/libraries/easyMesh-master/src/easyMeshSTA.cpp:13:
/home/andre/arduino-1.8.3/libraries/easyMesh-master/src/easyMeshSTA.cpp: In member function 'void easyMesh::tcpConnect()':
/home/andre/.arduino15/packages/esp8266/hardware/esp8266/2.4.0/tools/sdk/lwip2/include/lwip/ip4_addr.h:199:54: error: request for member 'addr' in '*(uint8*)(&((easyMesh*)this)->easyMesh::_stationConn.espconn::proto.espconn::<anonymous union>::tcp->_esp_tcp::local_ip)', which is of non-class type 'uint8 {aka unsigned char}'
 #define ip4_addr1(ipaddr) (((const u8_t*)(&(ipaddr)->addr))[0])
                                                      ^
/home/andre/.arduino15/packages/esp8266/hardware/esp8266/2.4.0/tools/sdk/lwip2/include/lwip/ip4_addr.h:205:38: note: in expansion of macro 'ip4_addr1'
 #define ip4_addr1_16(ipaddr) ((u16_t)ip4_addr1(ipaddr))
                                      ^
/home/andre/.arduino15/packages/esp8266/hardware/esp8266/2.4.0/tools/sdk/lwip2/include/arch/cc.h:87:24: note: in expansion of macro 'ip4_addr1_16'
 #define IP2STR(ipaddr) ip4_addr1_16(ipaddr), \
                        ^
/home/andre/arduino-1.8.3/libraries/easyMesh-master/src/easyMeshSTA.cpp:199:23: note: in expansion of macro 'IP2STR'
                       IP2STR(_stationConn.proto.tcp->local_ip),
                       ^
/home/andre/.arduino15/packages/esp8266/hardware/esp8266/2.4.0/tools/sdk/lwip2/include/lwip/ip4_addr.h:200:54: error: request for member 'addr' in '*(uint8*)(&((easyMesh*)this)->easyMesh::_stationConn.espconn::proto.espconn::<anonymous union>::tcp->_esp_tcp::local_ip)', which is of non-class type 'uint8 {aka unsigned char}'
 #define ip4_addr2(ipaddr) (((const u8_t*)(&(ipaddr)->addr))[1])
                                                      ^
/home/andre/.arduino15/packages/esp8266/hardware/esp8266/2.4.0/tools/sdk/lwip2/include/lwip/ip4_addr.h:206:38: note: in expansion of macro 'ip4_addr2'
 #define ip4_addr2_16(ipaddr) ((u16_t)ip4_addr2(ipaddr))
                                      ^
/home/andre/.arduino15/packages/esp8266/hardware/esp8266/2.4.0/tools/sdk/lwip2/include/arch/cc.h:88:5: note: in expansion of macro 'ip4_addr2_16'
     ip4_addr2_16(ipaddr), \
     ^
/home/andre/arduino-1.8.3/libraries/easyMesh-master/src/easyMeshSTA.cpp:199:23: note: in expansion of macro 'IP2STR'
                       IP2STR(_stationConn.proto.tcp->local_ip),
                       ^
/home/andre/.arduino15/packages/esp8266/hardware/esp8266/2.4.0/tools/sdk/lwip2/include/lwip/ip4_addr.h:201:54: error: request for member 'addr' in '*(uint8*)(&((easyMesh*)this)->easyMesh::_stationConn.espconn::proto.espconn::<anonymous union>::tcp->_esp_tcp::local_ip)', which is of non-class type 'uint8 {aka unsigned char}'
 #define ip4_addr3(ipaddr) (((const u8_t*)(&(ipaddr)->addr))[2])
                                                      ^
/home/andre/.arduino15/packages/esp8266/hardware/esp8266/2.4.0/tools/sdk/lwip2/include/lwip/ip4_addr.h:207:38: note: in expansion of macro 'ip4_addr3'
 #define ip4_addr3_16(ipaddr) ((u16_t)ip4_addr3(ipaddr))
                                      ^
/home/andre/.arduino15/packages/esp8266/hardware/esp8266/2.4.0/tools/sdk/lwip2/include/arch/cc.h:89:5: note: in expansion of macro 'ip4_addr3_16'
     ip4_addr3_16(ipaddr), \
     ^
/home/andre/arduino-1.8.3/libraries/easyMesh-master/src/easyMeshSTA.cpp:199:23: note: in expansion of macro 'IP2STR'
                       IP2STR(_stationConn.proto.tcp->local_ip),
                       ^
/home/andre/.arduino15/packages/esp8266/hardware/esp8266/2.4.0/tools/sdk/lwip2/include/lwip/ip4_addr.h:202:54: error: request for member 'addr' in '*(uint8*)(&((easyMesh*)this)->easyMesh::_stationConn.espconn::proto.espconn::<anonymous union>::tcp->_esp_tcp::local_ip)', which is of non-class type 'uint8 {aka unsigned char}'
 #define ip4_addr4(ipaddr) (((const u8_t*)(&(ipaddr)->addr))[3])
                                                      ^
/home/andre/.arduino15/packages/esp8266/hardware/esp8266/2.4.0/tools/sdk/lwip2/include/lwip/ip4_addr.h:208:38: note: in expansion of macro 'ip4_addr4'
 #define ip4_addr4_16(ipaddr) ((u16_t)ip4_addr4(ipaddr))
                                      ^
/home/andre/.arduino15/packages/esp8266/hardware/esp8266/2.4.0/tools/sdk/lwip2/include/arch/cc.h:90:5: note: in expansion of macro 'ip4_addr4_16'
     ip4_addr4_16(ipaddr)
     ^
/home/andre/arduino-1.8.3/libraries/easyMesh-master/src/easyMeshSTA.cpp:199:23: note: in expansion of macro 'IP2STR'
                       IP2STR(_stationConn.proto.tcp->local_ip),
                       ^
/home/andre/.arduino15/packages/esp8266/hardware/esp8266/2.4.0/tools/sdk/lwip2/include/lwip/ip4_addr.h:199:54: error: request for member 'addr' in '*(uint8*)(&((easyMesh*)this)->easyMesh::_stationConn.espconn::proto.espconn::<anonymous union>::tcp->_esp_tcp::remote_ip)', which is of non-class type 'uint8 {aka unsigned char}'
 #define ip4_addr1(ipaddr) (((const u8_t*)(&(ipaddr)->addr))[0])
                                                      ^
/home/andre/.arduino15/packages/esp8266/hardware/esp8266/2.4.0/tools/sdk/lwip2/include/lwip/ip4_addr.h:205:38: note: in expansion of macro 'ip4_addr1'
 #define ip4_addr1_16(ipaddr) ((u16_t)ip4_addr1(ipaddr))
                                      ^
/home/andre/.arduino15/packages/esp8266/hardware/esp8266/2.4.0/tools/sdk/lwip2/include/arch/cc.h:87:24: note: in expansion of macro 'ip4_addr1_16'
 #define IP2STR(ipaddr) ip4_addr1_16(ipaddr), \
                        ^
/home/andre/arduino-1.8.3/libraries/easyMesh-master/src/easyMeshSTA.cpp:201:23: note: in expansion of macro 'IP2STR'
                       IP2STR(_stationConn.proto.tcp->remote_ip),
                       ^
/home/andre/.arduino15/packages/esp8266/hardware/esp8266/2.4.0/tools/sdk/lwip2/include/lwip/ip4_addr.h:200:54: error: request for member 'addr' in '*(uint8*)(&((easyMesh*)this)->easyMesh::_stationConn.espconn::proto.espconn::<anonymous union>::tcp->_esp_tcp::remote_ip)', which is of non-class type 'uint8 {aka unsigned char}'
 #define ip4_addr2(ipaddr) (((const u8_t*)(&(ipaddr)->addr))[1])
                                                      ^
/home/andre/.arduino15/packages/esp8266/hardware/esp8266/2.4.0/tools/sdk/lwip2/include/lwip/ip4_addr.h:206:38: note: in expansion of macro 'ip4_addr2'
[... 9 lines cut ...]
 #define ip4_addr3(ipaddr) (((const u8_t*)(&(ipaddr)->addr))[2])
                                                      ^
/home/andre/.arduino15/packages/esp8266/hardware/esp8266/2.4.0/tools/sdk/lwip2/include/lwip/ip4_addr.h:207:38: note: in expansion of macro 'ip4_addr3'
 #define ip4_addr3_16(ipaddr) ((u16_t)ip4_addr3(ipaddr))
                                      ^
/home/andre/.arduino15/packages/esp8266/hardware/esp8266/2.4.0/tools/sdk/lwip2/include/arch/cc.h:89:5: note: in expansion of macro 'ip4_addr3_16'
     ip4_addr3_16(ipaddr), \
     ^
/home/andre/arduino-1.8.3/libraries/easyMesh-master/src/easyMeshSTA.cpp:201:23: note: in expansion of macro 'IP2STR'
                       IP2STR(_stationConn.proto.tcp->remote_ip),
                       ^
/home/andre/.arduino15/packages/esp8266/hardware/esp8266/2.4.0/tools/sdk/lwip2/include/lwip/ip4_addr.h:202:54: error: request for member 'addr' in '*(uint8*)(&((easyMesh*)this)->easyMesh::_stationConn.espconn::proto.espconn::<anonymous union>::tcp->_esp_tcp::remote_ip)', which is of non-class type 'uint8 {aka unsigned char}'
 #define ip4_addr4(ipaddr) (((const u8_t*)(&(ipaddr)->addr))[3])
                                                      ^
/home/andre/.arduino15/packages/esp8266/hardware/esp8266/2.4.0/tools/sdk/lwip2/include/lwip/ip4_addr.h:208:38: note: in expansion of macro 'ip4_addr4'
 #define ip4_addr4_16(ipaddr) ((u16_t)ip4_addr4(ipaddr))
                                      ^
/home/andre/.arduino15/packages/esp8266/hardware/esp8266/2.4.0/tools/sdk/lwip2/include/arch/cc.h:90:5: note: in expansion of macro 'ip4_addr4_16'
     ip4_addr4_16(ipaddr)
     ^
/home/andre/arduino-1.8.3/libraries/easyMesh-master/src/easyMeshSTA.cpp:201:23: note: in expansion of macro 'IP2STR'
                       IP2STR(_stationConn.proto.tcp->remote_ip),
                       ^`

Thanks

mjoelmendes, 2018-03-27:
Hello 
me too, what's the solution?

BlackEdder, 2018-03-31:
@mjoelmendes As far as I know easymesh has been abandoned (see the git activity). I am maintaining a fork though: https://gitlab.com/painlessMesh/painlessMesh which should compile fine (and if not, feel free to submit a bug report :))

rkoptev, 2018-04-03:
Having the same issue. Thank you, @BlackEdder , will try your library now.